dlaidig/vqf #2: External Acceleration of Moving Land Vehicle

goksanisil23, 2022-07-12:
Hi,
Firstly thank you for your extensive work and the care you've taken by providing implementations in different languages and a clear comparison against other open source methods with many datasets.

I would like to ask about the expected filter performance for the pitch and roll estimation of a moving land vehicle. The accelerometer measurements of such vehicle is frequently "disturbed" by the external accelerations due to the movement of the vehicle itself, which makes the usage of accelerometer's gravity vector based correction corrupted by such accelerations. I can see that the accelerometer values are low-pass filtered, however the fusion result I'm getting from the filter shows pretty much the same characteristic I get with other filters: Very high magnitude of roll & pitch at the presence of vehicle acceleration (braking/moving from standstill).

It would be great if you could give some insight in tuning the filter if it can handle such external accelerations. 

Thanks!

dlaidig, 2022-07-13:
Hi,

Thanks for your interest in VQF! All IMU orientation algorithms use the assumption that the direction of gravity can be recovered from the accelerometer measurements and, of course, that fails if there is a long phase of acceleration. So I am not too surprised that you observe those errors in your application. But I do have some ideas for how to mitigate them:

1. Increase `tauAcc` to extend the time window for which the accelerometers are taken into account so that the acceleration phase has a smaller impact. If the quality of the IMU measurements is good (and especially if the gyroscope bias is small or properly estimated by VQF), you should get away with significantly larger values without decreasing the overall accuracy too much.

2. Detect those acceleration phases and then disable the accelerometer update temporarily (just only call `updateGyr` or set the accelerometer measurements to [0 0 0]). Alternatively, you could also increase tauAcc only during those phases. 

3. If you have another way to know how the vehicle is accelerating, you could calculate the additional resulting acceleration vector that the IMU measures due to this acceleration and subtract this vector from the actual measurement. (That should work for any orientation estimation algorithm, of course.)

goksanisil23, 2022-07-23:
Hi, 
Thanks a lot for your thorough answer. Can confirm that increased tauAcc gives much more realistic results from the filter.

I have indeed thought of compensating for the base vehicle's acceleration, however we only have access to velocity measurements from which the derivation of acceleration gives a bit noisy result.

closed 2022-07-23 by goksanisil23