dlaidig/vqf #7: Quaternion Output not as Expected

kbhakt, 2022-10-27:
Thank you for creating a really nice resource of comparing different orientation filters and making it open-source. I am trying to use the VQF library in MATLAB and Python and am hoping you can help me make sense of some of the results I am seeing.

I am trying to use a custom IMU with a LSM9DS1 chip in order to get orientation data from the device. The device uses an ENU coordinate frame: x (East), y (North), and z(Up). This is the same as I understand as the reference frame used in paper (Fig. 2).

I have been able to create a visualizer in MATLAB to understand how my IMU moves in space using the MATLAB AHRS filter. This filter works but is a little bit delayed and not as accurate after several minutes, hence I am trying to use a different filter. (https://www.loom.com/share/6b1f63707f4a42d29519c9580e03cef3). 

I tried to implement the VQF filter (https://www.loom.com/share/34f7258738f84fc8950f0cbc782f405b), but noticed that I am not getting the sensor to be oriented correctly. I am confused since I am using the "ENU" frame reference for the AHRS filter and as I understand that is the default configuration for the VQF filter. 

Upon looking at the results, I noticed that it seems like my quaternion output values are swapped in certain locations. For example I have noticed that I have had to map from [w, x, y, and z] --> [y, z, w, x] (similar to what I saw here: https://github.com/dlaidig/vqf/issues/3). But this does not yield the same results as the MATLAB AHRS filter in all axes. I notice that one of the axes will rotate in the opposite direction of how I move the sensor.

I am wondering if there are any transformations I need to make to my input variables (I used the same acc, gyro, and mag inputs for both filters)? Are there any other considerations I should be aware of?

Any help in this matter would be greatly appreciated.

Snippet of MATLAB Code I am using:
I made sure that I was using the units mentioned in the documentation of rad/s, m/s^2, and mag could be arbitrary (I passed in uT).

        params = struct();
        params.motionBiasEstEnabled = false;
        params.restBiasEstEnabled = false;
        params.magDistRejectionEnabled = false;
        vqf = VQF(1/119, 1/119, 1/80, params);
        
        vqf.update(gyro_radspersec_Vec, accel_mps_Vec, mag_uT_Vec);
        temp_q = vqf.getQuat9D;
        % tried to change from [w, x, y, z] --> [y, z, w, x] in order to get right orientation in ENU frame (check VQF github issues tab)
        % temp_q2 = [temp_q(3), temp_q(4), temp_q(1), temp_q(2)];
        % temp_q3 = quaternConj(temp_q2);
        q = quaternion(temp_q);

dlaidig, 2022-10-28:
Hi Krishan,

I spent some time testing how `ahrsfilter` behaves. I am not really sure what Matlab does when you choose ENU and NED, but in the end I found out how to make the output quaternions compatible, so I didn't investigate any further.

I have documented the result on a newly created FAQ page in the documentation:
https://vqf.readthedocs.io/en/latest/faq.html#ref-faq-ahrsfilter

Can you please check if this helps solving your issues?

(Unrelated, I noticed that you use a different sampling time for the magnetometer measurements. I hope that means that you only pass `mag` in 80/119=67% of the update steps and otherwise just use `update(gyr, acc)`.)

kbhakt, 2022-10-28:
Hi Daniel,

Thanks for taking a look at this behavior. The coordinate transformation outline in the new FAQ page helped get me to the correct frame of reference -- thank you! I had some follow up questions that I was hoping you could answer:

1. Since each update step is modular in the update function (see below), why can't I pass in different sampling rates? Couldn't a delta time be computed separately for the mag vector, or is there something else happening/assumption I am missing? I am a bit confused since there is a way to set the magTs value in the updateMag function right? (magTs = obj.coeffs.magTs;)

        function update(obj, gyr, acc, mag)
            % Performs filter update step for one sample.
            %
            % :param gyr: gyr gyroscope measurement in rad/s -- 1x3 matrix
            % :param acc: acc accelerometer measurement in m/s² -- 1x3 matrix
            % :param mag: optional mag magnetometer measurement in arbitrary units -- 1x3 matrix
            obj.updateGyr(gyr);
            obj.updateAcc(acc);
            if nargin >= 4
                obj.updateMag(mag);
            end
        end

2. I tried using a different IMU sensor where I could specify the sampling rates to all be the same at 60 Hz. I was able to map to get the device to output correctly in my visualizer, but I noticed it took a while to return to the resting position after I rotated the sensor. It would take about 2-4 seconds before returning to the correct orientation. Would the parameters (tauAcc and tauMag) be good places to start tuning to get a faster response behavior.

3. Could you provide sample data for the acc, gyro, and mag you used to validate the quaternion outputs. I am worried that my sensor with different sampling rates may have noisy input data and would like to refer to something that has been validated to check for sign conventions, axes, and quality of the signal. I'm assuming that if the input data is noisy -- then orientation calculation could be as well.

4. Next, I am trying to implement this in Python. Is there a good way to visualize the output there? I have a way to send it to the matlab visualizer but I want to be able to have it agnostic from matlab if possible. I noticed you had a another github repo (https://github.com/dlaidig/qmt). Would this be a good place to start?

dlaidig, 2022-10-28:
Hi Krishan,

great to hear that this helped! :)

1. You can pass different sampling rates for gyr, acc, and mag. I was only pointing out that you then need to make sure that `updateMag` is also called with the correct frequency. If you only use `update(gyr, acc, mag)`, all 3 update functions will be called with the same frequency. If you specify a different sampling rate for mag, this implies that you should either only use `update(gyr, acc)` sometimes or call `updateGyr`, `updateAcc`, and `updateMag` separately.
2. You can use smaller values for `tauAcc` and `tauMag` to make the correction happen faster. However, if you get large disagreements after rotations that are not extremely fast, it is quite likely that there is an issue with the sensor data.
3. Here is the (not very optimized) code that I used. The IMU data can be found at https://github.com/dlaidig/broad/tree/main/data_mat.

```matlab
clc; clear; close all;
data = load('01_undisturbed_slow_rotation_A.mat');
gyr = data.imu_gyr;
acc = data.imu_acc;
mag = data.imu_mag;
sample_rate = data.sampling_rate;

f = ahrsfilter('SampleRate', sample_rate, 'ReferenceFrame', 'NED');
quat_ahrsfilter = f(acc, gyr, mag);

vqf = VQF(1/sample_rate);
out = vqf.updateBatch(gyr, acc, mag);
quat_vqf = quaternion([1/sqrt(2) 0 0 -1/sqrt(2)]) * quaternion(out.quat9D);

figure()
subplot(311)
plot(compact(quat_vqf))
title('adjusted VQF quaterion')
subplot(312)
plot(compact(quat_ahrsfilter))
title('ahrsfilter NED quaternion')
diffAngle = [];
for i=1:length(quat_vqf)
    diffAngle(end+1) = rad2deg(2*acos(abs(parts(conj(quat_ahrsfilter(i)) .* quat_vqf(i)))));
end
subplot(313)
plot(diffAngle)
title('difference angle [°]')

f = ahrsfilter('SampleRate', sample_rate, 'ReferenceFrame', 'ENU');
quat_ahrsfilter = f(acc, gyr, mag);

vqf = VQF(1/sample_rate);
out = vqf.updateBatch(gyr, acc, mag);
quat_vqf = quaternion([0 0 1 0]) * quaternion(out.quat9D);

figure()
subplot(311)
title('adjusted VQF quaterion')
plot(compact(quat_vqf))
subplot(312)
plot(compact(quat_ahrsfilter))
title('ahrsfilter ENU quaternion')
diffAngle = [];
for i=1:length(quat_vqf)
    diffAngle(end+1) = rad2deg(2*acos(abs(parts(conj(quat_ahrsfilter(i)) .* quat_vqf(i)))));
end
subplot(313)
plot(diffAngle)
title('difference angle [°]')
```

4. You might indeed find `qmt` to be helpful. I would suggest trying to get `examples/webapp_example_script.py` to work. This example shows the different options of how to use the visualization tools.

dlaidig, 2022-11-07:
Hi Krishan,

I'm closing this issue since, as far as I can tell, the original issue seems to be resolved. Feel free to re-open this issue or to create new ones.

closed 2022-11-07 by dlaidig