dlaidig/vqf #8: Problems importing the python vqf library

lorenzcsunikl, 2022-11-07:
Hi,
I installed the vqf library using pip and try to import it but I get the following error:

_import vqf
Traceback (most recent call last):

  File "/tmp/ipykernel_22215/1305596562.py", line 1, in <cell line: 1>
    import vqf

  File ".../lib/python3.8/site-packages/vqf/__init__.py", line 5, in <module>
    from .vqf import VQF, offlineVQF

  File "vqf/vqf.pyx", line 1, in init vqf.vqf

ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject_

As indicated I am using python 3.8 and numpy version 1.20.3

dlaidig, 2022-11-07:
Hi,

There was a [change in numpy 1.20.0](https://numpy.org/doc/stable/release/1.20.0-notes.html#size-of-np-ndarray-and-np-void-changed) that led to very similar errors. This is why numpy>=1.20.0 is now required in `setup.py`. I have not seen similar errors since then, but numpy 1.20.3 is also quite old (May 2021) and the combination of recently compiled vqf modules with older numpy versions was probably never tested.

The two things I would try are

1. upgrade numpy
2. alternatively, try to compile the C extension module yourself with the older numpy version, either with  `pip --no-binary :all:` (https://pip.pypa.io/en/stable/cli/pip_install/#cmdoption-no-binary) or by checking out the git repo and running `pip3 install .`

lorenzcsunikl, 2022-11-07:
Problem solved. Thank you very much for the quick reply!

closed 2022-11-07 by lorenzcsunikl