dlaidig/vqf #10: Non-rest bias calculation is poor

Quick-Flash, 2023-03-24:
I've written some code to just simulate some rotations on the VQF, and I was particularly interested in the non-rest bias calculations. What I found is that the non-rest bias calculations tend to only converge to the correct bias calculation when the IMU is in certain orientations. When you are flat and level yaw bias stays near 0 (which makes sense as the acc can't tell what is going on with yaw when you are level), but if you are tilting as you showed in the example code where the acc is reading 5.663 for each axis, we end up in a situation where the in flight bias is near 0 for each axis, even while it is at rest, but forced to not use the non rest calculations. I would imagine that in this orientation any gyro movement (roll, pitch or yaw) would result in changes to the acc and as such I would expect that the acc can learn the bias in this orientation, but the current method does not allow for it. The current non rest bias method tends to only converge to the correct bias when one or two acc axis are close to 0. When two are close to zero then it will calculate 2 bias, and when only 1 is close to zero it will calculate 1 bias.

This seems like a bug and not a feature of how the VQF acts. I could be entirely wrong, but it currently feels like the inflight calculation of bias could be improved to converge to the correct answer more often. I plan to use the VQF in a situation where you normally get only one chance to use the resting bias method. Its also not impossible for the imu I am using to change its drift over time (due to temperature changes), and would really like to see the non rest bias convergence work well at fixing some of these potential issues.

dlaidig, 2023-05-03:
Sorry for the late reply.

I'm not sure if I fully understand the observation you made, so I'll try to summarize what the motion bias estimation method can observe:

If the IMU is more or less stationary in the same orientation for a longer time (but the rest detection is not triggered), the horizontal component (i.e., two axes) of the gyroscope bias will be estimated, and the estimate for the vertical axis will slowly converge to zero. It is important to keep in mind that those axes do not need to correspond to the axes of the IMU coordinate system. In the example you mentioned with acc = [5.663, 5.663, 5.663], this means that the bias is not observable in [1, 1, 1] direction, while it is fully observable in the other two directions (e.g., [1, -0.5, -0.5] and [0, 1, -1], which are horizontal when the IMU is in this example orientation).

If the IMU assumes various different orientations, the horizontal direction always changes and over time, the estimate should converge for all axes.

> I would imagine that in this orientation any gyro movement (roll, pitch or yaw) would result in changes to the acc and as such I would expect that the acc can learn the bias in this orientation, but the current method does not allow for it.

In this orientation, a rotation around the [1 1 1] IMU axis (i.e., a combination of roll, pitch, and yaw) would not cause any change in the accelerometer measurement.

I hope this helped to clarify what the bias estimation method does. With this in mind, could you please describe in which situation the bias estimation failed and which observations led to this conclusion?

dlaidig, 2023-05-31:
I'm closing this for now. Feel free to re-open and to continue the discussion.

closed 2023-05-31 by dlaidig