dlaidig/vqf #12: There is a gap in the results

lxc5720, 2024-01-22:
hi @dlaidig Thanks for your great works and released code!
When I tried to use my own imu to collect data for testing, I found a gap between the algorithm using VQF and the real value, I assure you that my inputs meet the needs of your algorithm (e.g. m/s^2, rad/s, etc.), but I don't understand what causes this gap. I hope to get your reply.
[image]

dlaidig, 2024-01-22:
Thanks for your interest in VQF! Without further information, it's hard to come up with potential causes for the disagreement. Can you tell me a bit more about what you are testing and what your plot shows? For example:

- What is the ground truth and how did you obtain it?
- Which version of VQF did you use and did you adjust the parameters?
- Is the plot showing Euler angles? If so, how did you calculate them?
- What is the unit of the y-axis? Degrees?
- What is the sampling rate?

lxc5720, 2024-01-22:
I am very grateful to get your reply. Here is my answer to your question:
1. the real data comes from a 6-axis imu module which can output the Euler angles directly, I have tested the module and it is more accurate, so I use it as the real value in my comparison.
2. I used a version of matlab that did not change any of your default parameters during testing.
3.The graph I am showing is Euler angles, I am using quat2angle(out.quat6D,'XYZ') provided by Matlab to get the Euler angles.
4. yes, the y-axis is in degrees.
5. the sampling rate is 500Hz for both gyroscope and accelerometer.
I can provide the raw data if you want.Thanks again for your reply.

lxc5720, 2024-01-22:
I am very grateful to get your reply. Here is my answer to your question:
1. the real data comes from a 6-axis imu module which can output the Euler angles directly, I have tested the module and it is more accurate, so I use it as the real value in my comparison.
2. I used a version of matlab that did not change any of your default parameters during testing.
3.The graph I am showing is Euler angles, I am using quat2angle(out.quat6D,'XYZ') provided by Matlab to get the Euler angles.
4. yes, the y-axis is in degrees.
5. the sampling rate is 500Hz for both gyroscope and accelerometer.
I can provide the raw data if you need it. Thanks again for your reply

dlaidig, 2024-01-22:
Thanks! That already helps. Some thoughts:

1. The ground truth data seems to start perfectly at zero. I highly doubt that this is the result of aligning the IMU extremely precisely with gravity. So it seems like the IMU module already performs some kind of "reset", either manually or at the beginning. I would attempt to do something similar for the VQF orientations. For example, this should be possible with a loop over the data and something like `VQF.quatMultiply(VQF.quatConj(out.quat6D(1,:)), out.quat6D(i,:))`.
2. I am not sure how the Matlab `quat2angle` function works and I cannot know what your IMU module is doing, but I would try to see if changing the rotation order makes sense. At least try the reverse, 'ZYX', but maybe even systematically try all 12 possible rotation orders.

When you compare the Euler angles, you have to keep two things in mind:

1. The heading is undefined with 6D data, so one of the three angles (yaw) can have an arbitrary offset.
2. If the difference between the various rotation orders is not too clear, I suggest performing a movement with a lot more rotation in all directions. Then the influence of the rotation sequence should be a lot easier to see.

dlaidig, 2024-02-02:
Is there any update on this?

lxc5720, 2024-02-03:
Sorry, I just saw your message. I tried another data and replaced quatern2euler(out.quat6D) with quatern2euler(quaternConj(out.quat6D)). quaternConj solves for the conjugate of the quaternion, and in this kind of, achieves a similar output to the imu module.
[image]
[image]
[image]
[image]
Also, I used Mahony and Madgwick algorithms, and you can see that the output of Madgwick is closer to the output of the imu module.
For the original data, I tried as per your suggestion but still could not obtain similar output as imu module.

dlaidig, 2024-02-08:
Good to hear that the results for the second dataset look plausible and that my suggestion to invert the orientation seemed to have helped.

I don't know what kind of module you are using, but it's not unlikely that Madgwick's algorithm or something similar is used to estimate the orientation. If this is the case, it's not surprising that the obtained angles agree better. In order to really see which algorithm is more accurate, you would need a proper ground truth like marker-based optical motion capture.

For the first dataset, I would recommend to try to understand where the initial divergence of around 2° comes from and why the output of your IMU module does not have it. I'm quite sure you will be able to see this angle if you average the first few accelerometer measurements and calculate the angle between this vector and [0 0 1].

If you think that the raw data does not show this angle of around 2°, feel free to upload the raw accelerometer and gyroscope data and I will take a look.

closed 2024-02-08 by dlaidig