dlaidig/vqf #14: VQF C++ Implementation Issue with Acceptance of Reference/Candidate Magnetic Norm and Dip

dpar21, 2024-03-05:
Hello, 

I am attempting to utilize the VQF algorithm via the .cpp/.hpp files to run on a generic 9-axis IMU. However, the algorithm does not seem to run completely out of the box with the predominant issue being that of the acceptance portion of the updateMag() function. The adoption of a reference field seems to have a problematic effect on the quats that are returned from the updateGyro() and updateAcc() functions, which run before the updateMag() function when enabled for the 9D update, and are characterized by constant drift immediately following adoption. 

I have attached 2 graphs demonstrating the issue I am describing in more detail.

The first graph is that of the quaternion from the VQF updateGyro() function along with 2 parameters: the reference magnetic norm to show when acceptance has occured &  the gyroscope no bias estimate which is also affected with this acceptance: 

[image]

The second graph is that of the quaternion from the VQF updateAcc() function along with 2 parameters: the reference magnetic norm to show when acceptance has occured & again the gyroscope no bias estimate which is also affected with this acceptance: 

[image]

If you are able to provide any insight into why this may be occuring that would be much appreciated as the 9D update should be happening sequentially where any magnetometer updates should not affect the gyro or acc updates. 

Thank you, 

Dhyey Parikh

dlaidig, 2024-03-06:
This is indeed looking weird. But you are right: The magnetometer measurements and the magnetic field acceptance should not have any effect on `accQuat` and `gyrQuat`. I'm quite sure that you should be able to reproduce this without passing magnetometer measurements to VQF and that the root issue is something different.

What is the unit of GyrNoBias? Is this in rad or in m/s²? In any case, this is not something I've seen before. Can you check the norm of the original gyroscope measurements , gyroscope bias estimated by VQF, and the "rest detected" status?

dpar21, 2024-03-08:
Hello, 

Apologies for the delay! We actually seem to have found the issue in our implementation itself. It seems as if though when building the VQF algorithm onto the sensor and outputting the accepted reference magnetic field caused an issue with the accelerometer and gyroscrope quats for some odd reason. 

We took out the parameter when checking the original gyroscope measurements, gyro bias, and rest detected status as you suggested and it seems to work now. Thank you for your assistance! 

Dhyey Parikh

closed 2024-03-08 by dpar21