dlaidig/vqf #16: Execution speed on ARM Cortex-M4

larschristensen, 2024-05-31:
You mention in your paper that you were able to execute the VQF algorithm on an ARM Cortex-M4 at 1600 Hz. Could you clarify some details regarding
- Which version of the algorithm was executed? Was it 6D or 9D?
- What was the clock freqeuncy of the used ARM Cortex-M4?
- Was it the C++ code available in this repo that was executed?

dlaidig, 2024-06-03:
It was the full 9D version including magnetic disturbance rejection and rest detection + rest bias estimation. If I recall correctly, motion bias estimation was disabled but I didn't test if there would be performance issues when enabling this feature.

I believe the ARM Cortex-M4 was running at 64 MHz.

The code used was the C++ code available in this repo. VQF_SINGLE_PRECISION was enabled, i.e., double precision was only used for the Butterworth low-pass filters. (For which double precision is necessary, at least at high sampling rates like 1600 Hz, to avoid numerical issues.)

larschristensen, 2024-06-03:
Thanks for the info, closing.

closed 2024-06-04 by dlaidig