dlaidig/vqf #17: ValueError: numpy.dtype size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject

rmndrs89, 2024-06-17:
Hi Daniel,

I tried to install `vqf` via poetry in a new project on my Windows 10 Professional operating system, however it complains about some `numpy.dtype` error. I am not sure what is happening here, so I was hoping that you may have a clue?

```python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "D:\Projects\mimumove\.venv\lib\site-packages\vqf\__init__.py", line 5, in <module>
    from .vqf import VQF, offlineVQF
  File "vqf\vqf.pyx", line 1, in init vqf.vqf
ValueError: numpy.dtype size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject
```

To get there error, I did the following:

1. Start a new Windows Powershell session.
2. Navigate to the projects folder:  `cd D:\Projects`.
3. Start a new `poetry` project: `poetry new mimumove`.
4. Install Python libraries:
```bash
poetry add numpy scipy matplotlib
poetry add vqf qmt
```
5. Start a Python session: `python`
6. Import the libraries: `from vqf import VQF, BasicVQF, PyVQF`

I have also tried using a `venv` virtual environment rather than a `poetry` environment, but it gives the same error.
Is this something related to `vqf` or `numpy`?

Best regards,
Robbin

rmndrs89, 2024-06-17:
It may have to do with the new `numpy` [release](https://numpy.org/news/), I didn't realize this. Will try with a downgraded `numpy` version

dlaidig, 2024-06-18:
Note for myself: https://numpy.org/devdocs/dev/depending_on_numpy.html

dlaidig, 2024-11-15:
Hi Robbin,

Sorry that it took a while but I finally managed to find the time to update the Github actions, fix some minor issues, and publish a new version of vqf. Since the new wheels are built with numpy > 2.0, they should be compatible with both numpy 1.x and 2.x.

I've tested this on Linux but if you have a few minutes to test if vqf v.2.0.1 works for you, I'll appreciate some feedback.

closed 2024-11-15 by dlaidig

griffinmilsap, 2024-11-15:
Hey, thanks for addressing this!  I ran into this a few weeks ago and worked around it by capping my numpy version to pre-numpy 2.0 and I've upgraded and it works well!  Verified on MacOS/aarch64 with Python 3.10