dlaidig/vqf #19: Confusion about accelerometer correction ?

chennuo0125-HIT, 2024-09-21:
formula 4 in paper is the correction quaternion, as show below:
[image]
why is not it the following formula:
[image]

dlaidig, 2024-09-22:
You are digging deep. :smiley: 

Without more explanation, I cannot really tell where your equation comes from, but I did a quick numeric test and this shows that the equation in the paper yields unit quaternions while your version does not:

```pycon
>>> import numpy as np
>>> a = np.array([0.1, -0.2, 1])
>>> a = a / np.linalg.norm(a)
>>> q_w = np.sqrt((a[2]+1)/2)
>>> q_corr_paper = np.array([q_w, a[1]/2/q_w, -a[0]/2/q_w, 0])
>>> q_corr_paper
array([ 0.99395676, -0.09818335, -0.04909168,  0.        ])
>>> np.linalg.norm(q_corr_paper)
1.0
>>> q_corr_issue = np.array([q_w, a[1]*np.sqrt(1-q_w**2), -a[0]*np.sqrt(1-q_w**2), 0])
>>> q_corr_issue
array([ 0.99395676, -0.02142536, -0.01071268,  0.        ])
>>> np.linalg.norm(q_corr_issue)
0.9942453642140889
```

If we would numerically check that this also holds for any vector `a`, I would argue that this is enough to trust that the equation is correct because: (1) we know that $$q_w$$ is correct, (2) the axis of the quaternion is correct, and (3) the division by $$2 q_w$$ ensures unit norm.

Out of curiosity, I did some digging and found old notes. I'm happy to elaborate here what couldn't make it into the paper:

First of all, due to the properties of the cross product, the norm of the disagreement axis vector
```math
\mathbf{v}_{\mathrm{dis}} = [a_y  ~ -a_x ~  0]^T
```
is $\sin(\alpha_\mathrm{dis})$, where $\alpha_\mathrm{dis}$ is the disagreement (=correction) angle (i.e., the angle between the normalized acceleration vector $\mathbf{a}$ and $[0 ~ 0 ~ 1]^T$).

With this in mind, I hope the following derivation makes sense:

```math
\begin{align}
q_w &= \cos\left(\frac{\alpha_\mathrm{dis}}{2}\right) = \cos\left(\frac{\arccos(a_z)}{2}\right) = \sqrt{\frac{a_z+1}{2}}\\
q_x &= \sin\left(\frac{\alpha_\mathrm{dis}}{2}\right)  \frac{v_{\mathrm{dis},x}}{ \lVert \mathrm{v}_\mathrm{dis} \rVert} \\
&= a_y \frac{\sin\left(\frac{\alpha_\mathrm{dis}}{2}\right) }{\sin \alpha_\mathrm{dis}}\\
&= \frac{1}{2} \frac{a_y}{\cos\left(\frac{\alpha_\mathrm{dis}}{2}\right)}\\
&= \frac{1}{2} \frac{a_y}{q_w}
\end{align}
```

Let me know if you have any questions!

chennuo0125-HIT, 2024-09-22:
@dlaidig thanks for your reply, i know where mistake i made.  i forgot to normalize vdis :)

closed 2024-09-22 by chennuo0125-HIT