dlaidig/vqf #20: how to understand number 100 in code ?

chennuo0125-HIT, 2024-09-26:
many number 100.0 in setup() interface, how to understand it ?
[image]

dlaidig, 2024-09-26:
100 is a scaling factor to make sure the internal representation is in 0.01 °/s. 

See the documentation for `biasP` in https://vqf.readthedocs.io/en/latest/ref_cpp_state_coeffs.html#_CPPv4N8VQFState5biasPE:

> The 3x3 matrix is stored in row-major order. Note that for numeric reasons the internal unit used is 0.01 °/s, i.e. to get the standard deviation in degrees per second use $\sigma = \frac{\sqrt{p_{ii}}}{100}$.

IIRC, the numeric reason was related to the matrix inversion in the Kalman filter update, especially when using single-precision floats.

You can change this but I'm not sure why you would want to. In any case, you should carefully check the code using those variables to see if other places also need to be changed (e.g. `setBiasEstimate`).

chennuo0125-HIT, 2024-09-27:
thank you for your answer, i got it :) :) :)

closed 2024-09-27 by chennuo0125-HIT