dlaidig/vqf #24: How to handle dropped motion data?

griffinmilsap, 2025-02-05:
I'm working with a wireless device that reports gyro and accelerometer data at a regular sampling rate, but occasionally my wireless connection drops and I miss a small number of samples here and there.  Do you have any recommendations on how to handle these drops?   I found #9 which is quite a complicated feature request; perhaps this request is a bit less of an ask given the data is still at an underlying sampling rate.

Thankfully I have a sample number in my data stream so I know exactly how many samples I dropped, so I tried linearly interpolating the most recent value to the newest value in the incoming timeseries.  In practice, the orientation estimation suffers from a lot of drift when motion data is lost and linearly interpolated, especially without magnetometer fusion.  

I'm wondering if there's some more principled way to handle dropped or missing data; especially within the VQF module/algorithm itself.  I'm playing with the idea of applying a Kalman filter on the input to more accurately estimate instantaneous accelerometer/gyro data, but I'm concerned that this will smooth out a lot of the detail necessary to estimate orientation.

Also thanks for your time and effort developing and supporting this module!  It has aided my research greatly!

dlaidig, 2025-02-12:
If you cannot modify the sending part, I would also recommend to generate the missing samples by interpolation.

In this case, the main question is which interpolation strategy is best. Is it linear interpolation between the two samples, or is it better to repeat either the last sample before the gap or the first sample after the gap multiple times? Or are there more sophisticated interpolation strategies? Maybe it's worth investigating if there is some research about gyroscope interpolation.

If you can modify the sending part, there is a rather simple fix: Rather than transmitting the gyroscope data as a vector of 3 elements, integrate it (= `updateGyr` of `BasicVQF`) and then transmit the integral (= the `gyrQuat` quaternion).

On the receiving side, you can recover the gyroscope sample from the two quaternions and the sampling time `gyrTs` (see #22 for more details). If there is a gap and `N` samples are missing, you can do the same with `(N+1)*gyrTs` as the sampling time and then feed this gyroscope sample `N+1` times to VQF.

This should completely eliminate the drift you observed and only lead to artifacts during the gap. (And of course, there is also some influence of the gaps on the accelerometer-based correction -- but this is probably negligible.)

dlaidig, 2025-03-12:
I hope this answer was helpful. I'm closing this for now, but feel free to ask follow up questions.

closed 2025-03-19 by dlaidig

griffinmilsap, 2025-03-19:
With apologies for going cold on this, and many thanks for the prompt reply -- I had initially done a linear interpolation but I found a zero-order-hold interpolation (literally just repeat the last known value) produced equally erroneous but more usable results with less drift.

I believe the proper solution for this situation is to actually do this orientation estimation on the same device as the IMU and stream the estimated orientation vectors.   We've built our own device to do this, but our firmware is in Rust.  Thankfully I located a no_std rust crate with a port of your C++ implementation: https://crates.io/crates/vqf-rs. We'll give this a shot in the coming weeks :)

Thanks for the implementation!