dlaidig/vqf #25: VQF fails at high speed

swapnilsayansaha, 2025-03-04:
We are benchmarking VQF against different orientation filters using a Franka robotic arm. The arm provides ground truth quaternion and euler angles.

It seems that at high speeds, VQF's error explodes to as much as 40 degrees. At low speeds, this error is less than 1 degree for the same motion. The motion is a simple sinusoidal movement in one axis (either roll pitch or yaw)

Is there a parameter in VQF that needs to be tuned to prevent this?

dlaidig, 2025-03-04:
That sounds interesting.

Can you be a bit more specific about what "high speeds" means? What are the peak angular rates that you measure and how are the accelerations?

Is this behavior something where VQF stands out and has large errors while other common algorithms have much lower errors? Or do you observe similar errors with other algorithms?

In general, the main tuning parameter that you should look at is `tauAcc`.

However, before doing that I would recommend to look at the raw data and rule out any issues. First: Are you sure that there is no clipping in the gyroscope and accelerometer measurements at the high speeds?

swapnilsayansaha, 2025-03-04:
Hi. thanks for the prompt response. Settings of the franka robot are as follows:
• Angle : amplitude of the movement
  o 30 degrees
  o 120 degrees
• Speed
  o Low (0.33) - 22.83°/s
  o Medium (0.66) - 31.22 °/s
  o High (0.99) - 38.22°/s
We do not observe any clipping in accel or gyro measurements. We will try to play with tauAcc and come back to you.

swapnilsayansaha, 2025-03-05:
We have confirmed the units are fine and the sampling rate is also correctly specified. Tuning tauAcc helped reduce the absolute  error for high speed down to 7.5 degrees maximum for a 10 second log. The error happens due to time-varying drift, prevalent in yaw axis due to absence of magnetometer data.  The RMSE is 4.4 degrees for VQF as opposed to reported 2.9 degrees (still better than the 5.3-16.7 degrees RMSE for open-source attitude estimation filters).

[image]

closed 2025-03-05 by swapnilsayansaha