dlaidig/vqf #26: Is the quatRotate function correct?

szono, 2025-03-12:
I beg your pardon for this sudden enquiry.

I have read your paper and source code.
Before moving on to the main topic, I would like to express my gratitude for your hard works.
Thank you.

I read "Quaternions and Rotation Sequences" by J. B. Kuipers.
On page 126 of Kuipers' book, there are two quaternion operators expressed in the matrices Q and Q^t.
Comparing the two matrices and the quatRotate function written in vqf.cpp (line 728 to 734), they appear very different.
Here is a snippet of the function and an image that contains the matrices (sorry for the rough hand-writing).

```
void VQF::quatRotate(const vqf_real_t q[4], const vqf_real_t v[3], vqf_real_t out[3])
{
    vqf_real_t x = (1 - 2*q[2]*q[2] - 2*q[3]*q[3])*v[0] + 2*v[1]*(q[2]*q[1] - q[0]*q[3]) + 2*v[2]*(q[0]*q[2] + q[3]*q[1]);
    vqf_real_t y = 2*v[0]*(q[0]*q[3] + q[2]*q[1]) + v[1]*(1 - 2*q[1]*q[1] - 2*q[3]*q[3]) + 2*v[2]*(q[2]*q[3] - q[1]*q[0]);
    vqf_real_t z = 2*v[0]*(q[3]*q[1] - q[0]*q[2]) + 2*v[1]*(q[0]*q[1] + q[3]*q[2]) + v[2]*(1 - 2*q[1]*q[1] - 2*q[2]*q[2]);
    out[0] = x; out[1] = y; out[2] = z;
}
```

[image]

I assume the quatRotate function is intended to calculate the vector observed from the rotated coodinate system (i.e., frame).
If so, each element of the quatRotate function should match the one in Q^t, but it does not.

I would greatly appreciate if you could the two questions below.
1. Is the quatRotate function correct?
2. Which type of rotation is performed for each vector rotation, vector rotation or frame rotation?

This would be a great help as I am just starting to learn about IMU applications.
Thank you.

dlaidig, 2025-03-12:
Thanks for your interest in VQF!

I'm quite sure that the function is correct and does what it is supposed to do, but there are various conventions for orientations and rotations and the terminology is also often not used very consistently.

A few things that will hopefully help you:

1. quatRotate is implementing the following operation (as documented in https://vqf.readthedocs.io/en/stable/ref_cpp.html):

[image]

2. With the orientation estimate of VQF, this function can be used to obtain vectors in the global coordinate system when given vectors in the sensor's local coordinate system. (This is how VQF uses the function.)

3. Keep in mind that for unit quaternions, $q_0^2 + q_1^2 + q_2^2 + q_3^2 = 1$. From this follows $2q_0^2 +2 q_1^2 = 2 -  2q_2^2 - 2q_3^2$. With this, it is easy to show that the first element of your matrix is the same as what is used in quatRotate: $2q_0^2 +2 q_1^2 - 1 = 2 -  2q_2^2 - 2q_3^2- 1 = 1 -  2q_2^2 - 2q_3^2$. I'm quite sure the same can also be done for the other elements.
 
5. An even easier first check is probably to just numerically calculate the matrices for some random unit quternions. Then you'll directly see if the matrices are the same or if they are the transpose.

szono, 2025-03-13:
Thank you for your quick response and for the link to the reference and hints on why the rotation matrix is correct.

I also confirmed that the `quatRotate` function is correct.
I had a question why the `quatRotate` rotates the _vector_ in a fixed frame, rather than the frame ($qvq^{*}$, instead of $q^{*}vq$)
However, remembering that the accelerometer outputs the vector value that is opposite to the force vector applied to the sensor, $qaq^{*}$ actually transforms the acceleration vector $a$ to the one in a rotated frame.

I think that I see things clear now, but if you find any mistakes in the statement above, please let me know.

Thank you.

szono, 2025-03-18:
Thank you for your response.

I interpret the thumbs-up emoji as a confirmation that my statement is correct.
With this reply, I will close this issue.

Thank you again for your response and your time.

closed 2025-03-18 by szono