dlaidig/vqf #28: The calculation of the system noise variance of the gyroscope bias should consider the update frequency of the gyroscope data, right?

Eirenliel, 2025-03-28:
https://github.com/dlaidig/vqf/blob/a18fd32aa5d8c517f77a791a8357ffcfb093c71a/vqf/cpp/vqf.cpp#L965C45-L965C50

While working with vqf [we have noticed](https://github.com/SlimeVR/SlimeVR-Tracker-ESP/issues/370) that there is an error on this line, `accTs` is used where we're trying to categorize gyro, so `gyrTs` should be used instead.

Would love your comment on if that's really a mistake and it should be fixed, or it was made intentionally. Purely changing this to gyrTs gives us worse results, but it might be becase we tuned it with the error present.

cc: @Free-woodpecker @gorbit99 @kounocom

dlaidig, 2025-03-30:
After a quick look at the code, I'm quite sure that it is intentional and correct to use `accTs` to calculate `biasV` from the parameter `biasForgettingTime`.

The reason for this is that `biasV` is added to `biasP` inside the function `updateAcc`. This function is called with a sampling rate of `accTs`.

I agree that a physically accurate description of the process noise will mostly depend on the characteristics of the gyroscope measurements (which is very much affected by the actual gyropscope sampling rate). Still, since the Kalman filter update is performed at the accelerometer update, `accTs` would also be relevant to calculate a physically accurate value for `biasV`.

Since this is unrealistic in practice, VQF uses an alternative parametrization in which the `biasForgettingTime` describes how fast the uncertainty grows over time. See Appendix E of the paper for more details. For mapping this parametrization to `biasV`, only the accelerometer sampling time is relevant.

Eirenliel, 2025-03-30:
Thank you, that's a helpful explanation and I'm glad we didn't spend more time digging into it!

closed 2025-03-30 by Eirenliel